vhdl-linter/vhdl-linter #5: ranges are weird

g0t00, 2020-05-03:
ranges are weird

[image]

_Originally posted by @Nik-Sch in https://github.com/g0t00/vhdl-linter/issues/3#issuecomment-622978624_

g0t00, 2020-05-03:
Can not reproduce.
I have an idea. Maybe the file was not parsed successfully and you working on old snapshot at that time?

Nik-Sch, 2020-05-03:
Ah, I tested a bit and believe that it was cause by #3. I renamed the same signal twice. The first time it would remove the type of the signal and the parse cannot parse successfully and then ranges are wrong.

g0t00, 2020-05-03:
Found a different way to reproduce:
Introduce parser Error after entity port map. Include newline after entity.
Change port name.
Portname mention replaces produce error.

g0t00, 2020-05-03:
Added guard which disables renaming for invalid documents.

closed 2020-05-03 by g0t00