vhdl-linter/vhdl-linter #119: [Feature] Neovim support

pidgeon777, 2022-11-03:
First of all, congratulations on this project.

Is this plugin compatible with Neovim?

pidgeon777, 2022-11-03:
In VSCode usually, a LSP is started this way:

```
node C:\LSP\vscode-autohotkey2-lsp\server\dist\server.js --stdio
```

or

```
node C:\LSP\vscode-autohotkey2-lsp\server\dist\server.js --node-ipc
```

It could be possible to run your LSP in Neovim, I think.

Is there a particular command you're executing to start the LSP in VSCode?

g0t00, 2022-11-03:
So quite a while ago I testes this for the fun of it. It was working after some minor change, if I remember correctly.
Anyway after `npm ci` `npm run compile` you should be able to run `node dist/lib/language-server.js`. I think it is using node IPC for communication.
I currently dont have time to test myself, but this _should_ work.

g0t00, 2022-11-09:
So, I played around with this for a bit. I could not get it to run with node-ipc, but I created a branch which uses stdio instead of node-ipc. (nvim).
So, then nvim does not give any settings, so I made a small change on the nvim branch to fallback to the default config if none is given.
So, I did not test to much, but It seems to work kind of:
[image]
I am running out of time now, but i can make a command-line switch from node-ipc to stdio if that helps, and get that to the main branch.

pidgeon777, 2022-11-10:
Just switched to the `nvim` branch, generating the `language-server.js` file and using the `--stdio` node argument. I confirm the LSP now correctly starts.

I'll do some tests: shall I open a new issue reporting any found bug, or we can continue here?

Nik-Sch, 2022-11-10:
I would suggest to close this issue with #130 and then open separate issues for possible problems.

closed 2022-11-10 by pidgeon777

pidgeon777, 2022-11-14:
Another question, how could I pass the `vhdl-linter` LSP settings to the server in Neovim? This is what I tried:

```lua
...
init_options = {}
settings = {
  trace = {
    server = "verbose"
  },
  rules = {
    warnLibrary = true,
    warnLogicType = false,
    warnMultipleDriver = true,
  },
  style = {
    preferedLogicType = "std_logic",
    unusedSignalRegex = "_unused$"
  },
  ports = {
    enablePortStyle = false,
    outRegex = "^o_",
    inRegex = "^i_"
  },
  paths = {
    additional = {},
    ignoreRegex = ""
  },
  semanticTokens = false
},
...
```

g0t00, 2022-11-14:
So, I do not know how to do this. I have little experience with using neovim lsp support, maybe ask around at their github?
(settings is a standard feature of language server protocol).

Just a sidenote, I was lowkey thinking about providing a file based way to configure the linter, so you can add project specific stuff to git, but this will probably take a while..

pidgeon777, 2022-11-24:
> Another question, how could I pass the `vhdl-linter` LSP settings to the server in Neovim? This is what I tried:
>
> ```lua
> ...
> init_options = {}
> settings = {
>   trace = {
>     server = "verbose"
>   }, [...]

I solved that by wrapping the settings in a further `VhdlLinter` key:

```lua
    settings = {
      VhdlLinter = {
        trace = {
          server = "off" -- "off" | "messages" | "verbose"
        },
        rules = {
          warnLibrary = true,
          warnLogicType = false,
          warnMultipleDriver = true,
        },
        style = {
          preferedLogicType = "std_logic",
          unusedSignalRegex = "_unused$"
        },
        ports = {
          enablePortStyle = false,
          outRegex = "^o_",
          inRegex = "^i_"
        },
        paths = {
          additional = {},
          ignoreRegex = ""
        },
        semanticTokens = false
      },
    },
```