vhdl-linter/vhdl-linter #149: [Feature] Parse current file only

pidgeon777, 2022-11-24:
It seems that once a project root is found, each of the entities in the subfolders is scanned, leading to a slow LSP reactiveness in the case of many files.

It would be nice to add an option to parse only the active file, without scanning all of the project VHDL files.

g0t00, 2022-11-24:
I think this will be rather useless, as most VHDL files have dependency on other VHDL files (use clauses, instantiations etc.).
In theory you could make a reduced linting mode that does not wait on the parsing of the project, but it will be really limited in what it offers.
I don't see this as a priority, but feel free to propose a PR.

pidgeon777, 2022-11-24:
I think this could be useful for checking for potential issues affecting a single entity such as multiple driver signals, unused signals, etc., which pertain to the current entity architecture only, and not the other source/entity files.

In those cases of analyses, it wouldn't be needed to parse all of the project entities, just the current one.

Any hint on how the reduced linting mode could be added/enabled?

g0t00, 2022-11-24:
So, in langauge-server.ts the initialization Promise gets generated, which is awaited in all other ls functions. This among others waits on projectParser to parse the whole project.
Keep in mind that for the vhdl-linter access to ieee and standard libraries is also provided by the project parser, so without not even for example integer and std_logic_vector will be available.

And for your example of multiple driver/unused, if you have an instantiation in your entity the linter can not know if the ports are output or input thus can not if the signal is multiple driven.

Btw, do you know that in vhdl it is allowed (and people do it) to split entity and architecture into two files?

closed 2023-02-03 by Nik-Sch