vhdl-linter/vhdl-linter #201: u_unsigned not suggested

Nik-Sch, 2023-01-27:
Probably no alias is suggested for autocompletion

g0t00, 2023-01-27:
[image]
What do you mean?

Nik-Sch, 2023-01-27:
[image]
I mean the suggestions and not the quick fixes

Nik-Sch, 2023-02-01:
The same goes for record elements. With #200 merged, this should be possible now.
Similar with types that are defined in an interface_list

Nik-Sch, 2023-02-01:
reservedWords should also be suggested

Nik-Sch, 2023-02-01:
Aand interface formals could be suggested

Nik-Sch, 2023-02-07:
What is the status here? reserved words, interface formals and aliases are suggested at the moment? Should we have individual issues for the remaining problems?
Remaining would only be the record elements while typing which is problematic and I wanted to add parameters of subprograms in them.

g0t00, 2023-02-07:
for procedure parameters are suggested.
Currently function call is really missing, ass those are not currently parsed as instantiations...
This has multiple implications for stuff not working or not being check. This should become a new issue probably. Implementing auto complete is trivial then.
Yeah and maybe removing the wrong record elements from the auto complete becomes another separate issue.

Nik-Sch, 2023-02-07:
closing in favor of #235 and #236

closed 2023-02-07 by Nik-Sch