vhdl-linter/vhdl-linter #275: Parser stucked with configuration instance

cmeriaux, 2023-02-20:
When a module instantiate a sub-module with a configuration, parser is stucked
example 

    architecture rtl of my_module_wrapper is
    begin
        my_module_instance: configuration work.my_module
        port map(
            rstn      => rstn,
            clk       => clk,
            my_port_1 => my_port_1,
            my_port_2 => my_port_2
        );
    end rtl;

Here is a file to test the parser
[test.txt](https://github.com/vhdl-linter/vhdl-linter/files/10782204/test.txt)

By advance, Thank you

cmeriaux, 2023-02-20:
oups sorry, you are right, my test file was wrong
Should be like you said   `wrapper: configuration work.my_module_cfg  is `

Configuration is very powerfull, but could be tricky in some situations

g0t00, 2023-02-21:
Changes releases as [1.5.7](https://github.com/vhdl-linter/vhdl-linter/releases/tag/1.5.7)
If more problems arise feel free to open new issues.

commit 49a9c766 (2023-02-20): Parser stucked with configuration instance
Fixes #275
closed 2023-02-21 by g0t00