vhdl-linter/vhdl-linter #284: Can't parse alias of global signal declaration

cmeriaux, 2023-02-21:
in a testbench, using alias to replicate a nested signal is not recognized by the parser.
Here is an example:

`alias my_signal   is << signal u_dut.sub_module.my_nested_signal       : std_logic >>;`

Here is the parser error:

    signal 'u_dut' is referenced but not declared (not-declared)
    signal 'sub_module' is referenced but not declared (not-declared)
    signal 'my_nested_signal' is referenced but not declared (not-declared)

Alias syntax with << and >> could be tricky to parse !!!
Thanks

Nik-Sch, 2023-02-21:
Oh, haven't thought about this. Elaborating these correctly is going to be difficult because we would need to elaborate all the different files that are referenced in this expression. Currently only the opened file is elaborated and all other files are only parsed because elaboration can take quite long and is not feasible to do project wide on each key press.

Nik-Sch, 2023-02-23:
This actually should be solved by merging #273 when it is completed.

closed 2023-02-26 by Nik-Sch

cmeriaux, 2023-02-27:
Thank you guy for this mass quantity of rework to fix my issue.

commit b65354f7 (2023-02-24): add test for external names (and fix it)

fixes #284
commit 10136e6a (2023-02-26): Refactor reference elaboration (#273)

* wip: refactor elab-references

* fixed alias

* benchmarking

* Revert "benchmarking"

This reverts commit cac591859455eb635741e862222a2872dcac040a.

* selectedNames: lib.pkg + lib.pkg.obj

* selected names

* more selectedNames...

* Revert "more selectedNames..."

This reverts commit 24aa786a790d82908de250dabeb7f60a6aae2d64.

* package insts

* impossible to find selected names

* label links

* more package(inst) references

* references in package instantiations

* forgot some enums

* lint

* fix double elaboration

* kill elab use clause

* use cl [... cut]