vhdl-linter/vhdl-linter #286: [Feature] Add `vhdl-linter` project file

pidgeon777, 2023-02-21:
It would be great to add a project file to `vhdl-linter`, to enable or disable certain features, or to define the list of sources to be analyzed.

This [VHDL-Tool](https://www.vhdltool.com/) software uses a `vhdltool-config.yaml` project file with this format:

```yaml
#Define your project's libraries and source files here.
#This section is compulsory.
Libraries:
    #The name of the library.
    - name: hardware_lib
    #The paths where the source files for this library can be found. Use "**" to match arbitrarily nested directories.
      paths: 
        - "hardware_lib/**/*.vhd"
        - "hardware_lib/**/*.vhdl"

    - name: hardware_lib2
      paths: 
        - "hardware_lib2/**/*.vhd"
        - "hardware_lib2/**/*.vhdl"

    #Point to the IEEE standard libraries
    - name: ieee
      paths:
        - "/home/awesome-vhdl-dev/ieee/*.vhd"
        - "/home/awesome-vhdl-dev/ieee/*.vhdl"

Preferences:
    #Enable/disable typechecking
    TypeCheck: True

    #Display errors over multiple lines
    MultiLineErrors: True

    #Enable/disable check-as-you-type
    CheckOnChange: True

    #Enable/disable linting
    Lint: True

    #Display the first sintax error only in a file
    FirstSyntaxErrorOnly: True

#Linter rule configuration.
#Rules can be enabled or disabled.
#Rules also have a severity. It may be one of Info, Warning, Critical or Error.
Lint:
    #Threshold, below which messages are not displayed.
    Threshold: Warning

    #Long form rule configuration. Both enabled/disabled status and severity can be configured this way.
    DeclaredNotAssigned: 
        enabled:  True
        severity: Warning #Default severity Warning

    #Short form. Only enabled/disabled status can be specified. Severity is the default for the rule.
    DeclaredNotRead:             True #Default severity Warning
    ReadNotAssigned:             True #Default severity Critical
    SensitivityListCheck:        True #Default severity Warning
    ExtraSensitivityListCheck:   True #Default severity Warning
    DuplicateSensitivity:        True #Default severity Warning
    LatchCheck:                  True #Default severity Critical
    VariableNotRead:             True #Default severity Warning
    VariableNotWritten:          True #Default severity Warning
    PortNotRead:                 True #Default severity Warning
    PortNotWritten:              True #Default severity Critical
    NoPrimaryUnit:               True #Default severity Warning
    DuplicateLibraryImport:      True #Default severity Warning
    DuplicatePackageUsage:       True #Default severity Warning
    DeprecatedPackages:          True #Default severity Warning
    ImplicitLibraries:           True #Default severity Warning
    DisconnectedPorts:           True #Default severity Critical
    IntNoRange:                  True #Default severity Warning
```

closed 2023-02-21 by Nik-Sch

pidgeon777, 2023-02-21:
I would like to ask if the `ignoreRexex` option supports multiple patterns and if yes, how.

If not, could I open a new issue for implementing that feature?

Nik-Sch, 2023-02-21:
That is why it is a regex (regular expression). It has the possibility to match multiple elements with `|`. You can refer to [regex101](https://regex101.com/) to see what is possible.

pidgeon777, 2023-03-15:
I'm using this one:

`ignoreRegex = "^(?!.*ieee).*$"`

By doing so, I suppose every project file other than the currently edited and the library ones, will be skipped.

If not, how could I achieve that? Or, if you prefer, run `vhdl-linter` in a sort of single file mode.