vhdl-linter/vhdl-linter #289: [Bug] Cannot detect `when x"XXX"` condition in `case` statement

pidgeon777, 2023-02-23:
Sample code:

```vhdl
case cnt_lval_fldgen is
	when x"0014" =>
		data_int <= "0000000" & FIRST_CODE_N2(3);
	when "0100000000000001" =>
		data_int <= FOURTH_CODE_EAV_VALID_N1(2 downto 0) & FOURTH_CODE_EAV_VALID_N0 & "0";
	-----------------------------
	when others =>
		-- data_int <= DATA_FLDGEN;
		data_int <= "01100110";
end case;
```

[image]

g0t00, 2023-02-23:
Thank you for reporting. I think this is duplicate of #226
@Nik-Sch Can you confirm?
On the other hand the binary bitstring is working.

Nik-Sch, 2023-02-23:
Eh, yeah this is a duplicate of #226.
The reason why the bit string without 'x' is shown correctly is because we currently treat stringLiterals (`".*"`) and bitstring literals (`x".*"`) differently... This should definitely be streamlined

closed 2023-02-27 by Nik-Sch

pidgeon777, 2023-02-28:
Works good 👍.

Nik-Sch, 2023-02-28:
Nice to hear! Should also be released in 1.5.11 now.

PR #322 (2023-02-27): added whenTokens to when clause
closes #289
commit 6753d65c (2023-02-27): added whenTokens to when clause (#322)

closes #289
commit 21034e4c (2023-02-28): fix linking of attribute references to non-attribute-declarations (#321)

* fix linking of attribute references to non-attribute-declarations

* added whenTokens to when clause

closes #289

* Revert "added whenTokens to when clause"

This reverts commit a74104af8cc36358dc3a2bde0d4278966ef8338c.

* added the reverse case...

* timeout

---------