vhdl-linter/vhdl-linter #329: "use mypackage.all;" linter reports "expected . found ';'"

Qbort, 2023-03-01:
It appears there might be a bug in the linter (or misconfiguration on my side):

[image]

Nik-Sch, 2023-03-01:
Can you provide us with an MWE?

From the error message it looks like this use clause is not in a declartive part and, therefore, needs a library. e.g. `use work.wps_pkg.all;`

Qbort, 2023-03-06:
First of: thanks a lot for the amazing work with vhdl-linter. It is making my life so much easier!

Also, I apologize as I'm relatively new to HDL programming. 

Changing the line to `use work.wps_pkg.all;` indeed makes vhdl-linter happy. Thank you for that.

However, I should mention that Libero SoC seems to have no problem to compile the statement  `use wps_pkg.all;` even without `library work;` before it. I have been trying to figure out why that is but no luck - presumably it uses the `work` library when none is specified.

I am now not sure if this is a quirk of Libero SoC or vhdl-linter... Feel free to close if you don't think this needs to be fixed. The workaround is sufficient for my needs for now.

Nik-Sch, 2023-03-08:
ghdl is not able to compile this expression and imo this conforms with the LRM. Therefore, I would vote to not implement this and stay with the current behaviour. What are your thoughts @g0t00?

g0t00, 2023-03-10:
I am pretty confident that `use wps_pkg.all;` in a global context is not legal according to the LRM. (`library work;` should be not needed as this is implicit declared i think).
Anyway the error message could be more clear. @Nik-Sch The Error message probably already changed with your rework of elaboration?
What we could do is offer a solution when a use clause does not start with a library. (Just offer to use work. instead)

Nik-Sch, 2023-03-10:
No, this error message did not change because this comes from the prepare use clauses in the file parser and I didn't change it.
But I like the idea of providing a quick fix.

g0t00, 2023-03-10:
Okay, interesting. So I agree that `use library.all` is legal. But it might work in a different way than I at first thought.
I made the following example:
```vhdl
package test_pkg is
  constant test_constant : integer := 5;
end package;
---------

use work.all;
entity test_use_clause is
  port (
    test_input : integer_vector(test_pkg.test_constant - 1 downto 0);
    test_input2 : integer_vector(test_constant - 1 downto 0)
    );
end test_use_clause;
```
ran this through ghdl. According to ghdl `test_input` is a valid declaration `test_input2` is not. Which make sense.
I will look into how much work this is to fix...

Nik-Sch, 2023-03-20:
@g0t00 this has been implemented in #335?

g0t00, 2023-03-20:
@Nik-Sch Partially, 

>  What we could do is offer a solution when a use clause does not start with a library. (Just offer to use work. instead)
That is not implemented yet.

@Qbort Can you test if `use proasic3e.all;` is no working as expected?

Qbort, 2023-04-04:
Apologies for the delay. Yes, thank you - `use proasic3e.all;` now works.

`use my_pkg.all;`  still complains, but I am not as concern about that. I can just do `use work.my_pkg.all;`. Now the complaint is more explanatory though (see below) - I think this issue could be closed?

[image]

closed 2023-05-25 by Nik-Sch