vhdl-linter/vhdl-linter #340: Question Semantic symbol highlighting coloring

std-rubic, 2023-03-15:
Hi, the semantic symbol highlighting is imo another cool feature. Thank you for that. Is there a way to configure the colors ( for Input output ports, Signals, constants...) separate ?

Nik-Sch, 2023-03-15:
Our semantic tokens only provide semantic classes and properties for each token and the colors come from your color scheme. It is quite possible that your color scheme matches different semantic classes to the same color (in vscode you can execute the command (ctrl+shift+p) "Developer: Inspect Editor Tokens and Scopes" to see which token has which categories).
That being said we provide an additional setting to give inputs a different class from outputs which is false by default ("VhdlLinter.semanticTokensDirectionColoring"). 
[image]
Signals, variables, output ports etc. have the same coloring as they behave identical and constants, generics etc. should have a different coloring as they are read only. The setting mentioned above will make inputs to the same coloring as constants.

std-rubic, 2023-03-15:
OK nice I will adapt the color scheme.

closed 2023-03-15 by std-rubic