vhdl-linter/vhdl-linter #372: VhdlLinter server crashed for large project

poorren, 2023-04-28:
Hello guys,

The vhdl linter frequently crashed on my laptop. This is vivado project which contains more than 30 ipcore generated output. The design source module is about 50 files, I wonder if I could exclude the ipcore folder to lint, which may avoid the crash?

Here's the output log.
=============================
<--- JS stacktrace --->

FATAL ERROR: NewSpace::Rebalance Allocation failed - JavaScript heap out of memory
 1: 00007FF73AB17E56 node::Buffer::New+50054
 2: 00007FF73AB1805F node::OnFatalError+463
 3: 00007FF739F49B90 v8::internal::WebSnapshotDeserializer::object_count+864
 4: 00007FF73D801957 v8::CppHeap::CollectGarbageInYoungGenerationForTesting+55255
 5: 00007FF739FF3D9D v8::CppHeap::GetAllocationHandle+176397
 6: 00007FF739FEFC94 v8::CppHeap::GetAllocationHandle+159748
 7: 00007FF739FDFDE3 v8::CppHeap::GetAllocationHandle+94547
 8: 00007FF739FDDAEC v8::CppHeap::GetAllocationHandle+85596
 9: 00007FF739FDADE6 v8::CppHeap::GetAllocationHandle+74070
10: 00007FF73C136F7C v8::CppHeap::GetHeapHandle+69292
11: 00007FF73C1287B8 v8::CppHeap::GetHeapHandle+9960
12: 00007FF73C003044 uv_random+5722212
13: 00007FF739FAE1DC v8::CFunctionInfo::CFunctionInfo+332876
14: 00007FF73C3B95A6 cppgc::internal::WriteBarrier::DijkstraMarkingBarrierRangeSlow+35446
15: 00007FF73A1B6D8F v8_inspector::V8StackTraceId::ToString+126111
16: 00007FF73A26FD16 v8_inspector::V8StackTraceId::ToString+883750
17: 00007FF73A26B69B v8_inspector::V8StackTraceId::ToString+865707
18: 00007FF73A26B4FD v8_inspector::V8StackTraceId::ToString+865293
19: 00007FF739F8D607 v8::CFunctionInfo::CFunctionInfo+198775
20: 00007FF73C0FD1B2 v8::EmbedderStateScope::~EmbedderStateScope+204226
21: 00007FF73C10F4A3 v8::EmbedderStateScope::~EmbedderStateScope+278707
22: 00007FF73A0CC427 v8::internal::OSROptimizedCodeCache::TryGet+200487
23: 00007FF6BFECC93C 
[Info  - 1:14:40 PM] Connection to server got closed. Server will restart.
TypeError: Cannot read properties of undefined (reading 'parent')
    at new ObjectBase (c:\Users\575\.vscode\extensions\g0t00.vhdl-linter-1.6.1\dist\lib\parser\objects.js:160:19)
    at new OName (c:\Users\575\.vscode\extensions\g0t00.vhdl-linter-1.6.1\dist\lib\parser\objects.js:214:9)
    at ExpressionParser.innerConstraint (c:\Users\575\.vscode\extensions\g0t00.vhdl-linter-1.6.1\dist\lib\parser\expressionParser.js:426:28)
    at ExpressionParser.innerConstraint (c:\Users\575\.vscode\extensions\g0t00.vhdl-linter-1.6.1\dist\lib\parser\expressionParser.js:409:41)
    at ExpressionParser.parseConstraint (c:\Users\575\.vscode\extensions\g0t00.vhdl-linter-1.6.1\dist\lib\parser\expressionParser.js:391:33)
    at SubtypeIndicationParser.parse (c:\Users\575\.vscode\extensions\g0t00.vhdl-linter-1.6.1\dist\lib\parser\subtypeIndicationParser.js:76:131)
    at InterfaceListParser.getTypeDefinition (c:\Users\575\.vscode\extensions\g0t00.vhdl-linter-1.6.1\dist\lib\parser\interfaceListParser.js:168:109)
    at InterfaceListParser.parse (c:\Users\575\.vscode\extensions\g0t00.vhdl-linter-1.6.1\dist\lib\parser\interfaceListParser.js:108:70)
    at ComponentParser.parse (c:\Users\575\.vscode\extensions\g0t00.vhdl-linter-1.6.1\dist\lib\parser\componentParser.js:23:37)
    at DeclarativePartParser.parse (c:\Users\575\.vscode\extensions\g0t00.vhdl-linter-1.6.1\dist\lib\parser\declarativePartParser.js:59:98)
TypeError: Cannot read properties of undefined (reading 'parent')
    at new ObjectBase (c:\Users\575\.vscode\extensions\g0t00.vhdl-linter-1.6.1\dist\lib\parser\objects.js:160:19)
    at new OName (c:\Users\575\.vscode\extensions\g0t00.vhdl-linter-1.6.1\dist\lib\parser\objects.js:214:9)
    at ExpressionParser.innerConstraint (c:\Users\575\.vscode\extensions\g0t00.vhdl-linter-1.6.1\dist\lib\parser\expressionParser.js:426:28)
    at ExpressionParser.innerConstraint (c:\Users\575\.vscode\extensions\g0t00.vhdl-linter-1.6.1\dist\lib\parser\expressionParser.js:409:41)
    at ExpressionParser.parseConstraint (c:\Users\575\.vscode\extensions\g0t00.vhdl-linter-1.6.1\dist\lib\parser\expressionParser.js:391:33)
    at SubtypeIndicationParser.parse (c:\Users\575\.vscode\extensions\g0t00.vhdl-linter-1.6.1\dist\lib\parser\subtypeIndicationParser.js:76:131)
    at InterfaceListParser.getTypeDefinition (c:\Users\575\.vscode\extensions\g0t00.vhdl-linter-1.6.1\dist\lib\parser\interfaceListParser.js:168:109)
    at InterfaceListParser.parse (c:\Users\575\.vscode\extensions\g0t00.vhdl-linter-1.6.1\dist\lib\parser\interfaceListParser.js:108:70)
    at EntityParser.parse (c:\Users\575\.vscode\extensions\g0t00.vhdl-linter-1.6.1\dist\lib\parser\entityParser.js:26:37)
    at FileParser.parse (c:\Users\575\.vscode\extensions\g0t00.vhdl-linter-1.6.1\dist\lib\parser\fileParser.js:156:45)
TypeError: Cannot read properties of undefined (reading 'parent')
    at new ObjectBase (c:\Users\575\.vscode\extensions\g0t00.vhdl-linter-1.6.1\dist\lib\parser\objects.js:160:19)
    at new OName (c:\Users\575\.vscode\extensions\g0t00.vhdl-linter-1.6.1\dist\lib\parser\objects.js:214:9)
    at ExpressionParser.innerConstraint (c:\Users\575\.vscode\extensions\g0t00.vhdl-linter-1.6.1\dist\lib\parser\expressionParser.js:426:28)
    at ExpressionParser.innerConstraint (c:\Users\575\.vscode\extensions\g0t00.vhdl-linter-1.6.1\dist\lib\parser\expressionParser.js:409:41)
    at ExpressionParser.parseConstraint (c:\Users\575\.vscode\extensions\g0t00.vhdl-linter-1.6.1\dist\lib\parser\expressionParser.js:391:33)
    at SubtypeIndicationParser.parse (c:\Users\575\.vscode\extensions\g0t00.vhdl-linter-1.6.1\dist\lib\parser\subtypeIndicationParser.js:76:131)
    at InterfaceListParser.getTypeDefinition (c:\Users\575\.vscode\extensions\g0t00.vhdl-linter-1.6.1\dist\lib\parser\interfaceListParser.js:168:109)
    at InterfaceListParser.parse (c:\Users\575\.vscode\extensions\g0t00.vhdl-linter-1.6.1\dist\lib\parser\interfaceListParser.js:108:70)
    at ComponentParser.parse (c:\Users\575\.vscode\extensions\g0t00.vhdl-linter-1.6.1\dist\lib\parser\componentParser.js:23:37)
    at DeclarativePartParser.parse (c:\Users\575\.vscode\extensions\g0t00.vhdl-linter-1.6.1\dist\lib\parser\declarativePartParser.js:59:98)
TypeError: Cannot read properties of undefined (reading 'parent')
    at new ObjectBase (c:\Users\575\.vscode\extensions\g0t00.vhdl-linter-1.6.1\dist\lib\parser\objects.js:160:19)
    at new OName (c:\Users\575\.vscode\extensions\g0t00.vhdl-linter-1.6.1\dist\lib\parser\objects.js:214:9)
    at ExpressionParser.innerConstraint (c:\Users\575\.vscode\extensions\g0t00.vhdl-linter-1.6.1\dist\lib\parser\expressionParser.js:426:28)
[... 28 lines cut ...]
    at FileParser.parse (c:\Users\575\.vscode\extensions\g0t00.vhdl-linter-1.6.1\dist\lib\parser\fileParser.js:156:45)

<--- Last few GCs --->

[34408:00000190002E4000]    38437 ms: Scavenge 3926.5 (4056.1) -> 3921.2 (4057.4) MB, 4.3 / 0.0 ms  (average mu = 0.658, current mu = 0.579) allocation failure; 
[34408:00000190002E4000]    38484 ms: Scavenge 3927.5 (4057.4) -> 3921.7 (4057.4) MB, 12.7 / 0.0 ms  (average mu = 0.658, current mu = 0.579) allocation failure; 
[34408:00000190002E4000]    38543 ms: Scavenge 3927.8 (4057.4) -> 3922.3 (4057.4) MB, 17.3 / 0.0 ms  (average mu = 0.658, current mu = 0.579) allocation failure; 

<--- JS stacktrace --->

FATAL ERROR: NewSpace::Rebalance Allocation failed - JavaScript heap out of memory
 1: 00007FF73AB17E56 node::Buffer::New+50054
 2: 00007FF73AB1805F node::OnFatalError+463
 3: 00007FF739F49B90 v8::internal::WebSnapshotDeserializer::object_count+864
 4: 00007FF73D801957 v8::CppHeap::CollectGarbageInYoungGenerationForTesting+55255
 5: 00007FF739FF3D9D v8::CppHeap::GetAllocationHandle+176397
 6: 00007FF739FEFC94 v8::CppHeap::GetAllocationHandle+159748
 7: 00007FF739FDFDE3 v8::CppHeap::GetAllocationHandle+94547
 8: 00007FF739FDDAEC v8::CppHeap::GetAllocationHandle+85596
 9: 00007FF739FDADE6 v8::CppHeap::GetAllocationHandle+74070
10: 00007FF73C136F7C v8::CppHeap::GetHeapHandle+69292
11: 00007FF73C1354B5 v8::CppHeap::GetHeapHandle+62437
12: 00007FF73C35A412 v8::internal::OSROptimizedCodeCache::OsrOffsetsFor+717586
13: 00007FF6BFECC93C 
[Error - 1:15:22 PM] The VhdlLinter server crashed 5 times in the last 3 minutes. The server will not be restarted. See the output for more information.

g0t00, 2023-04-29:
So, there is some problem with memory leaks/usage right now. The version 1.6.1 already implemented some fixes which makes it much better, but seemingly the problem is not solved....
1.6.1 on the other hand also adds some basical verilog parsing to extract module interfaces.
Can you for testing disable this, and see if it becomes better?
Setting: `VhdlLinter.analysis.verilogAnalysis`

Alternative you can use `VhdlLinter.paths.ignoreRegex` to ignore the IP location and see if this helps...

poorren, 2023-05-01:
> 

Hello g0t00,
Disable the setting VhdlLinter.analysis.verilogAnalysis seems to work on my project, the memory still increase by about 4GB, but still in a limited way, not occupy the whole memory space. The cpu jump high occasionally, and edit responses a bit slow.

I also tried the VhdlLinter.paths.ignoreRegex, the effect is not very clear, I am not sure the path regexp is the correct setting.

closed 2023-08-14 by Nik-Sch