vhdl-linter/vhdl-linter #461: [BUG]: VHLD-Linter syntax-highlight not work, when starting vscode and opening vhd-file with errors

ungultig1, 2023-09-29:
Hello,

there is a bug I noticed.
It's easy to reproduce following these steps:

1. Open your project-Vhdl-file in vscode.
2. put following simple syntax-error:   After end of architecture:
`
... some vhdl code
....
end architecture rtl;

mySignal_out <= signal     --- A nonsense syntax error at end of file
`
3. Now, VHLD-Linter syntax-highlight still works.
4. Now close vsode and open vscode and your project-file again.
5. Resul: VHDL-Linter syntax-highlight wont work anymore.

--> This also leads to malfunctions in "Rename Symbol" and "Go to Definitions /Declaration" etc.

This issues occur when pasting some pseudo codes in VHDL and working on this.
Closing vscode and opening again forces you to make your VHDL-code synthesizable again, before VHDL-Linter can keep its work.

Thank you for fixing this issue.

Greetings,

Nik-Sch, 2023-10-02:
Hi,
yes this is a "known issue" which is pretty hard to fix. The underlying problem is, that in some cases we are not able to completely parse the file into a valid syntax tree in which case we cannot extract any symbols. For some features (like syntax highlighting) we then fallback to an old code version, hoping that most of the highlighting still is correct. However, for finding declarations/usages etc. this is not possible. The closing and re-opening of VS Code will also restart the vhdl-linter and with that, there are no old version of the syntax tree available which could be used for syntax highlighting and, therefore, nothing seems to work.

In some places we are able to predict a syntax error and continue to parse the file (like an extra comma at the end of a parameter list). However, if the file is too broken (like in your example with some random code outside of any design entity) we are not able to deduce a way to parse the file and do not elaborate anything of the file.

g0t00, 2023-10-02:
I fear that is not as simple as it sounds: Statements in vhdl are not line based. You can split your statement on as many lines as you want (or put multiple statements on one line).
Also syntax errors can hide the changing of scope. eg. if there is a syntax error in the port list the end of the port list can be missed and the parser is confused and is in the wrong parser mode for the rest of the file.
vhdl-linter already has some code to detected some common errors. (for example if you have a semicolon at the end of a port map this is shown as an error only but the parser recovers from this).
But sadly I think this needs to be done on a case-by-case basis.
In your example at the end of the file only design_units (entity/arch) or stuff like use clause is allowed. All of these begin with an keyword and vhdl-linter could probably better handle this. If I find some time I might improve this a bit.

closed 2023-10-08 by g0t00

g0t00, 2023-10-08:
@ungultig1 I included checking for your specific case. If you have other examples where you think, error handling should be possible feel free to open other issues.