vhdl-linter/vhdl-linter #491: Can't use ignoreFiles parameter

cmeriaux, 2023-11-12:
I can't use parameter VhdlLinter.paths.ignoreFiles. I've tried several pattern but It seems to not work:

```
 **/modules/externals/**/*.vhd  
 **/modules/externals
 **/modules/externals/**
```

The aim is no ignore a folder which is very big, parsing time consuming, and irelevant pour an integrator.
Would you have an idea ?
Thanks

Nik-Sch, 2023-11-13:
Thank you for the bug report. Yes I can currently reproduce this. I wanted to have a deeper look into the ignore settings anyways.
It seems that `ignoreFiles` is currently only applied to the filename instead of the path...
I will try to fix and test this thoroughly in the next days.

closed 2023-11-17 by Nik-Sch

Nik-Sch, 2023-11-17:
The fix has been published as 1.8.20.
The glob expression now applies recursively to directories and also to individual files, so all of your globs above should work fine.

cmeriaux, 2023-11-17:
Thank you very much guys for this new feature. You guys are so responsive. This is very much appreciated

cmeriaux, 2023-11-27:
Hello @Nik-Sch 
I have a corner case with my configuration with several folders opened in VSCode.

My folders contains: 
```
- module_A
- module_B
- module_C
```

I'd like to ignore  some folder in module_A and module_B but not in module_C. Here is my pattern:
```
- **/module_A/**/modules/externals/**
- **/module_C/**/modules/externals/**
```

VHDL Linter doesn't match the ignore rule because it uses a relavtive path to the workspace root. 
Would you have an idea how can I do achieve my needs ?
Thanks

Nik-Sch, 2023-11-27:
Ah, you are specifying the this in the global VS Code configuration? VS Code can have global settings, settings for the workspace and also settings for the individual folders. I believe you can overwrite the settings in `module_B`.
[image]

Additionally, you can also use our configuration file `vhdl-linter.yml` in each module to specify a module specific configuration that will also be applied to the cli version of the vhdl-linter.

cmeriaux, 2023-11-27:
I use the Workspace configuration which contains several folders. 
I'll give a try with the individual folder config .
Thanks

cmeriaux, 2023-11-27:
It seems configuration for individual folder is not available for vhdl-linter extension.
Do you confirm ?

Nik-Sch, 2023-11-27:
Ehh, yes you are correct... I will have a look into this.

cmeriaux, 2023-11-28:
hello @Nik-Sch  it's now fixed
Thank you very much