vhdl-linter/vhdl-linter #492: Ignore too large file

cmeriaux, 2023-11-12:
Some repository may contains large file like post synthesis netlist, which sizes more than hundred of Mega-Bytes.
That would be nice to have a parameter to ignore too large file ( size thresoldh would be configured by global settings)

thanks

Nik-Sch, 2023-11-13:
Currently there is a hard coded limit for VHDL files of 500KB which should not be parsed completely. However, in my opinion this limit is enforced in a too late stage of the parsing. (after checking for magic comments and some other random stuff in `lib/parser/fileParser.ts:119`.
We might want to have a limit that is additionally enforced in the projectParser to also avoid parsing large verilog files.
What do you think @g0t00?

g0t00, 2023-11-14:
@Nik-Sch Agreed, files ignored by size limitations should just be fully ignored. I was assuming that it what was happening. I would treat this current behaviour as bug.
As well for verilog files should just be ignored as well when to big.
While we are at it could make this configurable as well. But I think this is secondary.

closed 2023-11-17 by Nik-Sch

Nik-Sch, 2023-11-17:
The fix has also been published as 1.8.20.
There now is a setting (`analysis.maxFileSize`) which defaults to 512kiB which applies to vhdl and verilog files for the projectParser and also directly opening vhdl files.

cmeriaux, 2023-11-17:
Thank you very much guys for this new feature. You guys are so responsive. This is very much appreciated

PR #497 (2023-11-17): add setting for file size limit
This is now consistently enforced in the projectParser and also in the fileParser. fixes #492
commit 6d24aed6 (2023-11-17): add setting for file size limit (#497)

* add setting for file size limit

This is now consistently enforced in the projectParser and also in the fileParser.
fixes #492

* lint

* fix tests

* use async writeFile to fix mac?

* Apply suggestions from code review

---------