vhdl-linter/vhdl-linter #494: autocompletion of procedure fails

moertend, 2023-11-14:
If writing the name of the procedure followed by brace open, there no autocompletion list.
[image]

Nik-Sch, 2023-11-14:
I am pretty sure that this is the same problem as #493 as the file is currently not in a parsable state.

g0t00, 2023-11-15:
This is kind of not a duplicate, as here it has to be an instantiation. The othercase could also be recovered to be an assignment. The great think is that here we need already context information to deduce that foo( is not actually going to be somethink like `foo(1 downto 0) <= "10";`
Would like to keep this open, as it for me is about recovering from incomplete procedure instantiation.

Nik-Sch, 2023-11-15:
Yes you are correct, it is related but adds another level of recovery to it.