vhdl-linter/vhdl-linter #502: List of files instead of folder

Risto97, 2023-11-29:
Great tool!

Would you be willing to provide a way to pass a list of files instead of a folder for lining?
It is a more useful mode for integration into a build system.

Nik-Sch, 2023-11-29:
In theory this should be possible. However, a lot of vhdl code heavily relies on other files in the project (e.g. packages or instantiated entities) and the `vhdl-linter` has a so called project parser which first parses all vhdl files in the specified folder. After the parsing the files in the project, the files that are not excluded with the `-e` patterns are elaborated and linted to produce infos, warnings and errors.

To clarify your need, could you give me more information on how your project setup looks like and why it is required to just give specific files instead of a directory?

Risto97, 2023-11-29:
Sure I can explain.

I have a CMake based build system that has a way of describing hardware IP blocks and running EDA tools on them.
CMake recommendation and I follow it is to list all of your sources explicitly by listing them. Globbing (wildcard) is avoided!

For Verilog header files (I am not sure about VHDL as I am not an expert on it), you can add include_directories, and it will be added to the search path of the EDA tool.
```
add_ip(counter
    VENDOR github
    LIBRARY ip
    VERSION 0.0.1
    )

ip_sources(${IP} VHDL
    ${PROJECT_SOURCE_DIR}/file1.vhdl
    ${PROJECT_SOURCE_DIR}/file2.vhdl
    ${PROJECT_SOURCE_DIR}/counter.vhdl
    /tools/random_dir/asd.vhdl  # This should be avoided, but its allowed
    )

ip_include_directories(${IP}
    ${PROJECT_SOURCE_DIR}/inc
    )
```

So as you can see I have a very good idea of where my source files are, and I would like to avoid a tool (vhdl-linter) doing globbing on its own.

What would help me is if vhdl-linter had a mode that can receive a list of files, and a list of optional search directories.

Note that search path directories are tricky and some of the EDA tools require that apart from providing the search directory, you also need to pass the file path to the file list.