vhdl-linter/vhdl-linter #503: vhdl-linter.yml file has no effect

yotarid, 2023-12-07:
Hi, 

In your readme you claim that "Rules and style settings can be configured with vhdl-linter.yml configuration files which modifies the default settings".
In my case, I would like to use the `vhdl-linter` terminal command. However, including a `vhdl-linter.yml` file doesn't seem to make any difference. The rules and styles applied are still default ones (different from those configured in my VSCode workspace).
So what is needed in order for the `vhdl-linter` command to pick the rules and style defined in my `vhdl-linter.yml` configuration file ?

Thanks in advance

Nik-Sch, 2023-12-07:
And how and from where do you invoke the `vhdl-linter` executable?
I agree with you, that the settings file should be used. Maybe you could also provide a minimal working example where it is apparent that the settings file is not used.

yotarid, 2023-12-07:
Wonderful! works like charm now. 

Many thanks for your help! Closing the issue

closed 2023-12-07 by yotarid