vhdl-linter/vhdl-linter #508: [Help] Managing the `UNISIM` library

pidgeon777, 2024-04-02:
As the title says, an example showing how the Xilinx `UNISIM` library can be referenced in a project analyzed with `vhdl-linter` would be very appreciated.

The Xilinx `UNISIM` source files are usually available in the following folder:

`C:\Xilinx\Vivado\2023.2\data\vhdl\src\unisims`

So, I suppose the source files in it should be added to the `vhdl-linter` project file 
somehow?

Nik-Sch, 2024-04-08:
The `paths.additional` field in the settings file is a list of paths that can be used for external libraries:
```yml
paths:
    additional:
        - 'C:\Xilinx\Vivado\2023.2\data\vhdl\src\unisims'
```

closed 2024-04-08 by Nik-Sch

pidgeon777, 2024-04-08:
Thanks for the answer @Nik-Sch.

This is my setting:

```
paths:
    additional:
        - 'C:\Xilinx\Vivado\2023.2\data\vhdl\src\unisims\lite'
```

`lite` is a subfolder I created which contains the following Xilinx (untouched) files:

- unisim_retarget_VCOMP.vhd
- unisim_VPKG.vhd

`vhdl-linter` returns this error in the source file:

```vhdl
library UNISIM
use UNISIM.vcomponents.all; --> object 'vcomponents' is referenced but not declared (not-declared)
```

I wonder what's missing here. Maybe related to this?:

https://github.com/vhdl-linter/vhdl-linter/issues/507

NOTE: I didn't use the "original" `C:\Xilinx\Vivado\2023.2\data\vhdl\src\unisims` because otherwise `vhdl-linter` would crash (too much memory usage indexing the `unisim` entities).

Nik-Sch, 2024-04-09:
After looking further through the code, I just realized that the additional paths are only read from the VSCode settings and not from the `vhdl-linter.yml` file and with that also does not work with the command line. I neither have a windows system nor the Xilinx unisims at hand, however, I verified that the VSCode additional paths setting works.

At the moment there is no easy way to use the `vhdl-linter.yml` file for this setting and I am also not sure if it would make sense to use this setting from the file. The idea of the vhdl-linter.yml file was to have a file to be checked into a VCS for consistent style across the repo and an absolute path for any external libraries probably won't match on any other system