vhdl-linter/vhdl-linter #510: Lint only the file, which is open in vscode

ungultig1, 2024-04-15:
Hi there,

I was wondering if there is an option where I can force vhdl-linter to lint only the file that is open within vscode.

At the moment I am using vhdl-linter mainly for semantics, over a project conatining several vhdl files.

Unfortunately, the semantic feature of vhdl-linter stops working when opening between these multiple files;
because vhdl-linter always tries to parse all the vhdl files it finds when I open another file.

PS.: Could there also be an option to use the semantic feature only? 

Thank you.

Nik-Sch, 2024-04-15:
This probably does not make too much sense, because without the project parser almost nothing will work. (e.g. types like `std_logic` are defined in ieee packages and require the project parser to parse the files)

You could add the following to your `vhdl-linter.yml` to ignore all your files for the project parser:
```yml
paths:
  ignoreFiles:
    - "**/*"
    - ".**/*"
```
However, this will lead to a lot of "errors" being raised in the files you have open as a lot of objects will be referenced that have not been defined. You can then add
```yml
rules:
  - not-declared: false
```
to ignore all `not-declared` errors but this will obviously ignore all errors when you use some undefined signal.

ungultig1, 2024-04-15:
Thank you for your anwser.

I am using the "settings.json" of vscode to provide settings. 
Is this correct?:

```   
"VhdlLinter.paths.ignoreFiles": [
        "\"**/*\", \".**/*\""
    ],
```

Nik-Sch, 2024-04-16:
No, it should be
```json
"VhdlLinter.paths.ignoreFiles": [
        "**/*", ".**/*"

    ],
```